AMSAT-NA/PacSatSW #25: Time message error

billreed, 2023-10-01:
TIme message sends 9 digits, should be 10 digits.

ac2cz, 2023-10-02:
Thanks for the report.  Do you see the error message in the Pacsat console or do you mean it sends NO -5 to the ground station (Ill formed Packet error).

The time command works here of course.  I assume you are running the ground station on Linux?

ac2cz, 2023-10-02:
I have probably messed something up in the ground station and the Linux version is sending something different.  Here is what I get when I send the TIME command from my Windows laptop.

When PACSAT receives the TIME command I receive 67 bytes like this:

RX Bytes: 67:
ac 8a 64 a8 86 a0 f6 8e 60 96 98 82 40 61 3 bc cd ab be ad de 1a 0 1 e 0 ab d0 1a 65 0 0 0 0 1 35 33 4c 47 8e cd 90 4e 8c 4e 99 47 5b e8 29 99 fe b3 97 92 6a da 54 52 b5 0 64 2f f8 a1 62 d1                                                 

Everything up to 0xbc in byte 16 is the packet header which gives:             
RX- UI: G0KLA>VE2TCP-11 Cmd pid:bc pf:0 |len:51| 

So there are 51 data bytes as follows:
First 18 bytes for the command:
cd ab be ad de 1a 0 1 e 0 ab d0 1a 65 0 0 0 0  
with the time stamp hard coded with reset 0xABCD and uptime 0xDEADBE.  Little endian of course and then followed by Address 0x1A, Special 0x00, Name Space 0x01, Command 0x0E and 4 args each 2 bytes, which is this case are 0xd0ab, 0x651a, 0x00, 0x00

0x651ad0ab is 1696256171 - the unix time

Then 32 bytes for the hash:
1 35 33 4c 47 8e cd 90 4e 8c 4e 99 47 5b e8 29 99 fe b3 97 92 6a da 54 52 b5 0 64 2f f8 a 1 

Finally 2 bytes for the packet checksum:
62 d1

ac2cz, 2023-10-02:
Bill said he sees this:

Setting unix time to: 1696278401
TX: - UI: KD5GCT-11>TIME-1 Res pid:f0 pf:0 |len:10| 31 36 39 36 32 37 38
34 31 0

I read the original error and thought it was the time Command, but you clearly said Time MESSAGE, so my mistake.

I see now that it is only sending 9 bytes so this is an error.

Really this should be just 4 bytes.  The 32 bit number for the time.  I implemented this quickly so the ASCII value of the time would print in the ground station.  But we should send 4 bytes and translate it on the ground into the actual time.  We use this issue to keep track of this until it is fixed.

ac2cz, 2023-10-04:
This is resolved in version X0.1k and Ground Station version 0.45e

closed 2024-03-08 by ac2cz