AMSAT-NA/PacSatSW #37: We should try the RTC multiple times at startup before assuming it is not working

ac2cz, 2024-05-05:
We try to init the RTC at startup by resetting it and setting the clock bit.  If that fails we ignore it and use the unix time.  But any glitch on the I2C bus would cause this error.  Also at startup many things could be using the I2C bus and the call only blocks for a short while before failing.

We should try the init 3-5 times perhaps and then fail.  We should block uploads if we fail until the time is set.

If the time is set from the ground we should try to set the RTC and the Unix time but it is critical that the unixtime is set even if the RTC does not respond.

N5BRG, 2024-05-05:
When you speak of RTC you are talking about the routine in the TMS570 that maintains the current UTC time I assume. This routine would query the MAX31331 clock chip on the board.  This routine must be working at some level or "get rtc" would not be returning the proper UTC time on the satallite terminal.

I have tried the set time button on the ground station. I assume this reads the computer time and uses this to set the time on the satallite?

The time I see on the ground station is supposed to be the current time on the satellite?

I plan now to cycle through the "set unix time xxxx" on a restart of the satallite and set it at an odd time. Then try the set time again from the ground station.

I don't think the issue would be i2c bus related. If we really need to check this I have an Aardvark which can be used to monitor and log i2c traffic. Takes some time to setup so I will put this off for now.

ac2cz, 2024-05-05:
When I refer to the RTC I mean the max 31331 chip.  I will post more comments later but two things are important. 
1. The time on the spacecraft is maintained by RTOS and not by the RTC.  We just use the RTC to initialize it.
2. The ground commands are validated using the time, so if you play around moving the time on your ground station computer then it may ignore your commands.

N5BRG, 2024-05-06:
Ground station must do a set time which is getting the time from the satallite to set the time in the ground station for proper coordination. So this should be one of the first things you do when starting the ground station.

I have seen this working but I realize there is a sequence you must go though when starting up the satallite and the ground station and I need to do it a few more times so I can document it at least for myself. Not an issue when in orbit I hope.

ac2cz, 2024-05-06:
The ground station knows the time.  It should use the Network Time Protocol (NTP) or another mechanism to set it accurately.  This is part of Windows/Linux etc.

The satellite reads the time at boot from its Real Time Clock (RTC) but it might be wrong or that chip might have failed.  So it sends the time to the ground in a TIME packet.  If that time is wrong then the ground station can send a set time packet (perhaps automatically).  The ground station uses its own current time (in UTC) and sets the time on the satellite.  

If the RTC is broken then the time on the satellite will not survive a reboot and would have to be set continually.

Uploading files requires the clock to be accurate.  Otherwise files are stored in the past.  Managing expired files also requires an accurate clock but if the clock is not set then no files are expired.

ac2cz, 2024-05-06:
So there are several improvements that could be made here.
1. Why do we call GetStatus31331() and what purpose does it serve vs InitRtc31331()?
2. Do both startup routines need to succeed for the clock to be operational?
3. What happens if it fails, what do we do other than ignore the RTC?
4. Do we store the status in RTCStat and use that to prevent uploads and file expiry when the clock is not working?